joltwallet/esp_littlefs #179: Commit written data automatically

domnulvlad, 2024-02-25:
I am very much aware that LittleFS does not actually write any data to flash until the file is closed, or until sync functions are called. This is obviously a concern for power losses, especially considering this file system is designed to be resilient to such events.

I need to write data to a file for logging, and the system's power may be interrupted at any point. My solution is to set a timer, and so every 250ms I close and reopen the file. Sometimes, this procedure takes longer than usual, which introduces unwanted hiccups.

Have any advances been made towards fixing this issue? Is it not possible to have it automatically write data to disk once the RAM buffer is filled to, let's say, the size of a LittleFS block? I don't understand why the user must save data manually.

BrianPugh, 2024-02-25:
Have you seen/tried my comment here: https://github.com/joltwallet/esp_littlefs/issues/144#issuecomment-1732174884

Similarly, have you tried setting `CONFIG_LITTLEFS_FLUSH_FILE_EVERY_WRITE`?

BrianPugh, 2024-02-25:
Correct, see [this comment](https://github.com/espressif/esp-idf/issues/2820#issuecomment-446051814) detailing fflush/fsync.

> Also, https://github.com/littlefs-project/littlefs/issues/344#issuecomment-1226901862 mentions padding prior to fsync-ing to ensure all data is commited.

I think that padding was misunderstanding fflush/fsync and would implicitly trigger a `fflush`. A `fflush` followed by a `fsync` is required to clear all buffers and commit the writes to disk.

> So, after enabling it, I still have to call fflush after every little fprintf?

Correct, since `fflush` flushes buffers upstream of `esp_littlefs`. However, with `CONFIG_LITTLEFS_FLUSH_FILE_EVERY_WRITE` enabled, you **will not** have to additionally call `fsync`. Note that there will (not sure how practically significant) be a performance penalty for constantly flushing.

domnulvlad, 2024-02-25:
Then, regarding my initial question, it is not possible for a "commit to disk" to happen automatically and only when necessary (i.e. when enough bytes have been stored by fprintf/fwrite/etc. that could be written in a LittleFS block without padding with garbage)?

I'm not completely sure, but from my understanding, calling fflush (and fsync) every time a few bytes of data are stored would cause a commit that needs padding to reach the size of a block, which wastes time. Wouldn't it be more efficient for the system to just commit data when the size of a block is surpassed? That would definitely be more elegant than calling those functions after every write, or on a timer.

BrianPugh, 2024-02-25:
data is automatically committed to disk as needed; you only need to fflush/fsync/fclose to ensure that the file is in a "good state" that can then be read from again after you lose power.

BrianPugh, 2024-02-25:
I believe we are miscommunicating around two different things:

1. LittleFS (and, by extension, `esp_littlefs`) automatically performs erases/writes (distinct from a flush!) as necessary to flush caches in memory to make room for more data in the memory buffers.

2. Explicit `fsync` (calls the underlying `lfs_file_sync`). This flushes the `littlefs` buffers (independent and downstream from higher level filesystem buffers in the stack) to disk [**and manually updates the file's metadata on disk**](https://github.com/littlefs-project/littlefs/issues/943#issuecomment-1950289853). This metadata essentially encodes information like "how long the file is" and "where all the chunks are on disk".

Expanding on (2), the point of LittleFS is that the files and filesystem are *never in an inconsistent state*. I.e. the filesystem won't become corrupt from abrupt power loss. The explicit `fsync` calls tells littlefs "I want to commit to this file state to revert to if power is loss." And then of course, you should call `fflush` before `fsync` so all the data you think has been written gets flushed down to the LittleFS layer.

domnulvlad, 2024-02-26:
Alright. So if I'm using `fprintf` to write data, from [here](https://github.com/espressif/esp-idf/issues/2820#issuecomment-446051814) I understand I have to call `fflush` to transfer the C layer buffers to the LittleFS buffers, to make LittleFS "know" about the data I just wrote.

Then, if `CONFIG_LITTLEFS_FLUSH_FILE_EVERY_WRITE` is **disabled**, if I don't call `fsync`, then the data is still written to disk, but the metadata isn't updated, so after a power cycle the system won't know about the file?

domnulvlad, 2024-02-26:
Ok, I did some testing myself.

Just `printf`-ing in a loop takes 0ms to execute and, as expected, after reboot the file doesn't contain anything. After many prints, "lfs.c:689:error: No more free space 0x172" appears, which I assume means some buffer is full because it didn't get committed to flash.

Adding an `fflush` after every `printf` (after enabling `CONFIG_LITTLEFS_FLUSH_FILE_EVERY_WRITE`) makes the first few writes take 1-5ms, and then the time grows to a semi-steady 50-90ms, with the occasional "hiccup" of 150ms. Of course, after rebooting, the file contains data... But these times are completely unacceptable, especially when the rest of the program has to do other tasks in the order of milliseconds. In this case, I would say my method of periodically closing/flushing the file based on a timer is less time-intensive. Is there really no fix for this?

BrianPugh, 2024-02-26:
Ok, so this is more or less behaving as expected. Upstream LittleFS has performance issues when it comes to appending to a file (or similar). I'd recommend bringing this up in the official repo. This repo's focus is only on the glue between esp-idf's virtual filesystem and LittleFS.

domnulvlad, 2024-03-01:
With the default configuration but `CONFIG_LITTLEFS_FLUSH_FILE_EVERY_WRITE` enabled, the data is committed automatically to flash every 4096 bytes.

Every write takes 0ms, except for the one that has to commit, which takes ~80ms. I can't really complain, since the original prospect was 80ms EVERY write :)

closed 2024-03-01 by domnulvlad