joltwallet/esp_littlefs #181: How to prevent assert failed: lfs_bd_prog lfs.c:262 (pcache->block == ((lfs_block_t)-1)) when writing bigger fails e.g. greater than 64 kB?

spikeyamk, 2024-03-06:
```
    char buf[BUFSIZ];
    int create_test_file(const size_t size_bytes, const std::string_view& name) {
        const std::filesystem::path test_megabyte { std::filesystem::path(mount_point).append(name) };

        FILE* file { std::fopen(test_megabyte.c_str(), "w") };
        std::setbuf(file, buf);
        if(file == nullptr) {
            return -1;
        }

        char c { 'a' };
        for(
            size_t i = 1;
            i <= size_bytes;
            [&i, &c]() {
                if(c == 'z') {
                    c = 'a';
                } else {
                    c++;
                }

                i++;
            }()
        ) {
            if(std::fputc(c, file) != c) {
                return static_cast<int>(i);
            }
            if((i % BUFSIZ) == 0) {
                std::fflush(file);
            }
        }

        std::fclose(file);
        return 0;
    }
```
I'm doing something like this to create a test file with ESP32-C6 and 2GB SD card over single lane SPI. I've tried adding the std::fflush and std::setbuf for the FILE* stream and it doesn't seem to help at all. What is the actual maximum file size? And can you create files like 1MB in size and larger? Or not?

Logs:

```
I (10729) esp_littlefs: SD card formatted!
Success: SD_Card::format(): '0'
Success: SD_Card::create_test_file(unsigned int: '1', unsigned int: 'jcb1'): '0'
Success: SD_Card::check_test_file(unsigned int: '1', unsigned int: 'jcb1'): '0'
Success: SD_Card::create_test_file(unsigned int: '2', unsigned int: 'jcb2'): '0'
Success: SD_Card::check_test_file(unsigned int: '2', unsigned int: 'jcb2'): '0'
Success: SD_Card::create_test_file(unsigned int: '4', unsigned int: 'jcb4'): '0'
Success: SD_Card::check_test_file(unsigned int: '4', unsigned int: 'jcb4'): '0'
Success: SD_Card::create_test_file(unsigned int: '8', unsigned int: 'jcb8'): '0'
Success: SD_Card::check_test_file(unsigned int: '8', unsigned int: 'jcb8'): '0'
Success: SD_Card::create_test_file(unsigned int: '16', unsigned int: 'jcb16'): '0'
Success: SD_Card::check_test_file(unsigned int: '16', unsigned int: 'jcb16'): '0'
Success: SD_Card::create_test_file(unsigned int: '32', unsigned int: 'jcb32'): '0'
Success: SD_Card::check_test_file(unsigned int: '32', unsigned int: 'jcb32'): '0'
Success: SD_Card::create_test_file(unsigned int: '64', unsigned int: 'jcb64'): '0'
Success: SD_Card::check_test_file(unsigned int: '64', unsigned int: 'jcb64'): '0'
Success: SD_Card::create_test_file(unsigned int: '128', unsigned int: 'jcb128'): '0'
Success: SD_Card::check_test_file(unsigned int: '128', unsigned int: 'jcb128'): '0'
Success: SD_Card::create_test_file(unsigned int: '256', unsigned int: 'jcb256'): '0'
Success: SD_Card::check_test_file(unsigned int: '256', unsigned int: 'jcb256'): '0'
Success: SD_Card::create_test_file(unsigned int: '512', unsigned int: 'jcb512'): '0'
Success: SD_Card::check_test_file(unsigned int: '512', unsigned int: 'jcb512'): '0'
Success: SD_Card::create_test_file(unsigned int: '1024', unsigned int: 'jcb1024'): '0'
Success: SD_Card::check_test_file(unsigned int: '1024', unsigned int: 'jcb1024'): '0'
Success: SD_Card::create_test_file(unsigned int: '2048', unsigned int: 'jcb2048'): '0'
Success: SD_Card::check_test_file(unsigned int: '2048', unsigned int: 'jcb2048'): '0'
Success: SD_Card::create_test_file(unsigned int: '4096', unsigned int: 'jcb4096'): '0'
Success: SD_Card::check_test_file(unsigned int: '4096', unsigned int: 'jcb4096'): '0'
[... 59 lines cut ...]
riscv32-esp-elf-addr2line -pfiaC -e /home/spikeyamk/.espressif/tools/esp-rom-elfs/20230320/esp32c6_rev0_rom.elf 0x4003071e: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40813c80: 0x00000000 0x40813ef4 0x40811818 0x4003079a 0x00000000 0x00000000 0x00000000 0x40811818
riscv32-esp-elf-addr2line -pfiaC -e /home/spikeyamk/.espressif/tools/esp-rom-elfs/20230320/esp32c6_rev0_rom.elf 0x4003079a: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40813ca0: 0x00000000 0x00020000 0x40811818 0x42007672 0x37303630 0x39303830 0x00017b00 0x72313231
riscv32-esp-elf-addr2line -pfiaC -e /home/spikeyamk/esp/ESP32_AD5933/fw/build/cpp_rtti.elf 0x42007672: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40813cc0: 0x40815734 0x00000011 0x00000011 0x31323032 0x33323232 0x35323432 0x4081587c 0x408157ec
40813ce0: 0x00000011 0x0000001e 0x00647261 0x37333633 0x39333833 0x00000000 0x40813cb8 0x40813cbf
40813d00: 0x37343634 0x40813d84 0x00000014 0x420080d8 0x35353435 0x37353635 0x00000009 0x40813d8c
riscv32-esp-elf-addr2line -pfiaC -e /home/spikeyamk/esp/ESP32_AD5933/fw/build/cpp_rtti.elf 0x420080d8: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40813d20: 0x33363236 0x35363436 0x37363636 0x39363836 0x31373037 0x33373237 0x35373437 0x37373637
40813d40: 0x39373837 0x31383038 0x00000009 0x40813d8c 0x00000009 0x40813d8c 0x00000009 0x40813d8c
40813d60: 0x00000000 0x40813e24 0x00000014 0x00000000 0x00000000 0x40813e24 0x00000014 0x42008256
riscv32-esp-elf-addr2line -pfiaC -e /home/spikeyamk/esp/ESP32_AD5933/fw/build/cpp_rtti.elf 0x42008256: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40813d80: 0x00020000 0x40813d8c 0x00000009 0x3162636a 0x37303133 0x00000032 0x40811748 0x40813da4
40813da0: 0x00000009 0x3162636a 0x37303133 0x40810032 0x4080f1d0 0x4080f000 0x40813dc0 0x00000006
40813dc0: 0x30313331 0x42003237 0x00000001 0x4riscv32-esp-elf-addr2line -pfiaC -e /home/spikeyamk/esp/ESP32_AD5933/fw/build/cpp_rtti.elf 0x42003237: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
080f554 0x00000000 0x4080f000 0x42061000 0x4200645a
riscv32-esp-elf-addr2line -pfiaC -e /home/spikeyamk/esp/ESP32_AD5933/fw/build/cpp_rtti.elf 0x4200645a: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'
40813de0: 0x00000000 0x00000000 0x00000001 0x00008000 0x00000000 0x00000000 0x00000001 0x4080b7c2
riscv32-esp-elf-addr2line -pfiaC -e /home/spikeyamk/esp/ESP32_AD5933/fw/build/cpp_rtti.elf 0x4080b7c2: [Errno 2] No such file or directory: 'riscv32-esp-elf-addr2line'

ELF file SHA256: 26849f8d5

Rebooting...
```

BrianPugh, 2024-03-06:
[The line in question](https://github.com/littlefs-project/littlefs/blob/f53a0cc961a8acac85f868b431d2f3e58e447ba3/lfs.c#L262), where `lfs_block_t` is `uint32_t`.

I haven't seen this before. Theoretically, LittleFS should be able to handle files up to size ~2GB. Not to unnecessarily drag @geky in, but do you have any high level quick thoughts on this? Do we believe a bug lies in this repo (esp_littlefs) or in upstream official littlefs?

spikeyamk, 2024-03-06:
> [The line in question](https://github.com/littlefs-project/littlefs/blob/f53a0cc961a8acac85f868b431d2f3e58e447ba3/lfs.c#L262), where `lfs_block_t` is `uint32_t`.
>
> I haven't seen this before. [...]

Any idea where I should start digging? I've tried changing a bunch of settings in the esp_little configuration in sdkconfig it didn't really help either

BrianPugh, 2024-03-06:
IDK if this is possible in your setup, but can you try something similar with SPI flash? That then may point to if it's a generic problem, or an sd-card specific problem.

spikeyamk, 2024-03-06:
> IDK if this is possible in your setup, but can you try something similar with SPI flash? That then may point to if it's a generic problem, or an sd-card specific problem.

So, I used the same function as [here](https://github.com/joltwallet/esp_littlefs/issues/181#issue-2172008319)
created this in partitions.csv:
```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,        data, nvs,      0x9000,  0x6000,
phy_init,   data, phy,      0xf000,  0x1000,
factory,    app,  factory,  0x10000, 2M,
graphics,   data, spiffs,         ,  0xF0000, 

```
And the thing worked fine, it worked for 512 kB file, but at 1MB file it failed with:
`E (656363) esp_littlefs: ./managed_components/joltwallet__littlefs/src/littlefs/lfs.c:689:error: No more free space 0x1a5`
But that's a completely different error and says it ran out of space I use an 8MB FLASH ESP32-C6 devkit.

BrianPugh, 2024-03-06:
With those partitions (assuming the `graphics` partition, that has a size of `0xF0000 == 983040` bytes) that error makes sense.

So this is actually kind of good news; that means that the error is (probably, most likely) in this repo, esp_littlefs, and is isolated to the SD code. @huming2207 do you think you would be able to look into this?

spikeyamk, 2024-03-06:
So, the funny thing is:
when I do something like this in a loop:
```
    size_t integrity_test(const size_t max_bin_pow) {
        for(size_t i = 1; i < (2 << max_bin_pow); i *= 2) {
            const std::string name { std::string("jcb").append(std::to_string(i)) };
            if(Trielo::trielo<create_test_file>(Trielo::OkErrCode(0), i, name) != 0) {
                return i;
            }
            if(Trielo::trielo<check_test_file>(Trielo::OkErrCode(0), i, name) != 0) {
                return i;
            }
        }
        return 0;
    }

```
Where it creates a file and then creates a file twice as large as the iteration before it will fail at filesize 128 kB, but when I only do this once in app_main
```
extern "C" void app_main() {
	if(Trielo::trielo<SD_Card::init>(Trielo::OkErrCode(0)) != ESP_OK) {
		return;
	}
	if(Trielo::trielo<SD_Card::format>(Trielo::OkErrCode(0)) != ESP_OK) {
		return;
	}
	const std::string_view name { "test" };
	if(Trielo::trielo<SD_Card::create_test_file>(Trielo::OkErrCode(0), (2 << 18), name) != 0) {
		return;
	}
	if(Trielo::trielo<SD_Card::check_test_file>(Trielo::OkErrCode(0), (2 << 18), name) != 0) {
		return;
	}
	Trielo::trielo<SD_Card::deinit>(Trielo::OkErrCode(0));
}
```
and hardcode the value of the file size and create only one file, it will fail at file size 512 kB with:
`assert failed: lfs_bd_prog lfs.c:262 (pcache->block == ((lfs_block_t)-1))`

huming2207, 2024-03-07:
Hmmm, this looks like an SPI driver issue to me. See: https://github.com/espressif/esp-idf/issues/7804

Maybe try SDMMC interface and see if the issue still occurs?

huming2207, 2024-03-07:
By the way, what's the SD card sector size and sector count? @spikeyamk

spikeyamk, 2024-03-07:
> Hmmm, this looks like an SPI driver issue to me. See: [espressif/esp-idf#7804](https://github.com/espressif/esp-idf/issues/7804)
> 
> Somehow you might just transmitted more than 32KB for a single transaction underneath and cause some sort of corruption? Maybe try SDMMC interface and see if the issue still occurs? I will have a closer look later today.

Alright, I'll try later in the evening when I come back home...

> By the way, what's the SD card sector size and sector count? @spikeyamk

So this function: [sdmmc_card_print_info](https://github.com/espressif/esp-idf/blob/master/components/sdmmc/sdmmc_common.c#L265) gives me:

```
Name: 00000
Type: SDSC
Speed: 20.00 MHz (limit: 20.00 MHz)
Size: 1910MB
CSD: ver=1, sector_size=512, capacity=3911680 read_bl_len=10
SSR: bus_width=1
```

spikeyamk, 2024-03-07:
> Maybe try SDMMC interface and see if the issue still occurs? I will have a closer look later today.

So it seems like only ESP32S3 and ESP32 supports this and I only have a ESP32C6 board, so I can't do that...

geky, 2024-03-07:
Can confirm this is unlikely to be a littlefs issue. littlefs also doesn't support cross-block/cross-sector writes at the moment so that seems a bit unlikely if I understand the integration correctly.

Honestly I'd be a bit suspicious of the c++ stdlib/integration here. What is BUFSIZ? Could it be that the stdlib was compiled with a different BUFSIZ?

spikeyamk, 2024-03-07:
> Can confirm this is unlikely to be a littlefs issue. littlefs also doesn't support cross-block/cross-sector writes at the moment so [espressif/esp-idf#7804](https://github.com/espressif/esp-idf/issues/7804) seems a bit unlikely if I understand the integration correctly.
> 
> Honestly I'd be a bit suspicious of the c++ stdlib/integration here. What is BUFSIZ? Could it be that the stdlib was compiled with a different BUFSIZ?

BUFSIZ is 128 but honestly it don't matter. I added that thing with char buf[BUFSIZ] with std::setbuf(file, buf) and std::flush(file) as a desparate attempt to fix this and it failed the exact same way before.

huming2207, 2024-03-07:
> So I guess the sector size is 512 and total sectors 3911680.

This is similar to our 512 byte sector SD card (those XTX SD NAND chip). It works fine for us...

Also, could you please tell me what's the ESP-IDF version are you using? Could you please try v5.2 or newer if you were using the old ones?

spikeyamk, 2024-03-08:
> > So I guess the sector size is 512 and total sectors 3911680.
> 
> This is similar to our 512 byte sector SD card (those XTX SD NAND chip). It works fine for us...
> 
> Also, could you please tell me what's the ESP-IDF version are you using? Could you please try v5.2 or newer if you were using the old ones?

I'm using ESP-IDF 5.2

spikeyamk, 2024-03-08:
No other version of ESP-IDF seems to fix this. Also I found another error:
`E (27560) esp_littlefs: ./managed_components/joltwallet__littlefs/src/littlefs/lfs.c:1366:error: Corrupted dir pair at {0x22dc5, 0x22dc6}`
When creating a file with
`FILE* file = std::fopen("file_name", "w");
std::close(file);`
and then writing 512 bytes to the file and immediately closing it.
Then after few iterations of this in a loop it will fail at creating 74th file like this with the error above.

Also I tried mounting the using the sdspi API to write to SD Card blocks directly and it seems to work fine. I tested the card in Windows for bad blocks.

I also looked into code in this repo and here: [write to a partition](https://github.com/joltwallet/esp_littlefs/blob/master/src/littlefs_esp_part.c#L29) and [write to an sdcard](https://github.com/joltwallet/esp_littlefs/blob/master/src/littlefs_sdmmc.c#L31) function and the other function completely ignores the
`lfs_off_t off`
argument passed to the function

spikeyamk, 2024-03-08:
Also on this [line](https://github.com/joltwallet/esp_littlefs/blob/fd64733cdf248c7a7eb207db7d28124f8857fe0b/src/littlefs_sdmmc.c#L22) you do some weird thing instead of using the lsf_config_t* c function parameter passed to the function

BrianPugh, 2024-03-08:
> I also looked into code in this repo and here: [write to a partition](https://github.com/joltwallet/esp_littlefs/blob/master/src/littlefs_esp_part.c#L29) and [write to an [...]

It's not ignored, it's used immediately when computing `part_off`.

The corrupted dir pair at this point is somewhat expected, because it seems like data is being corrupted somewhere on the way to disk. It's to be expected that the read back data is also bad.

> Also on this [line](https://github.com/joltwallet/esp_littlefs/blob/fd64733cdf248c7a7eb207db7d28124f8857fe0b/src/littlefs_sdmmc.c#L22) you do some weird thing instead of using the lsf_config_t* c function parameter passed to the function

That `efs` comes from `lfs_config_t` a few lines above:

```
esp_littlefs_t * efs = c->context;
```

spikeyamk, 2024-03-09:
Ok, so I tried a bunch of different things:

1. I looked more closely at the code of esp_littlefs, but got confused by all this wrapping sdmmc API around littlefs and then wrapping littlefs around VFS API and then wrapping it around stdio and using `std::fopen`, `std::fwrite` etc. So I wrote this:
```
#include <cstdint>
#include <cstring>
#include <cstdio>
#include <cassert>
#include <array>
#include <string>

#include <littlefs/lfs.h>
#include <trielo/trielo.hpp>

#include "lfs_port/sdmmc/test.hpp"

namespace LFS {
    namespace SDMMC {
        int read(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, void *buffer, lfs_size_t size) {
            assert(off == 0);
            assert(size == c->block_size);
            assert(block < c->block_count);
            sdmmc_card_t* card = static_cast<sdmmc_card_t*>(c->context);
            assert(sdmmc_read_sectors(card, buffer, block, 1) == ESP_OK);
            return LFS_ERR_OK;
        }

        int erase(const struct lfs_config *c, lfs_block_t block) {
            assert(block < c->block_count);
            sdmmc_card_t* card = static_cast<sdmmc_card_t*>(c->context);
            assert(sdmmc_erase_sectors(card, block, 1, SDMMC_ERASE_ARG) == ESP_OK);
            return LFS_ERR_OK;
        }

        int prog(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, const void *buffer, lfs_size_t size) {
            assert(off == 0);
            assert(size == c->block_size);
            assert(block < c->block_count);
            sdmmc_card_t* card = static_cast<sdmmc_card_t*>(c->context);
            assert(sdmmc_write_sectors(card, buffer, block, 1) == ESP_OK);
            return LFS_ERR_OK;
        }

        int sync(const struct lfs_config *c) {
            (void) c;
            return LFS_ERR_OK;
        }

        int test(sdmmc_card_t* card) {
            const lfs_size_t block_size { 512 };
            const lfs_size_t block_count { static_cast<lfs_size_t>(card->csd.capacity) };
            static uint8_t read_buffer[block_size];
            static uint8_t prog_buffer[block_size];
            static uint8_t lookahead_buffer[block_size];

            const lfs_config config {
                .context = static_cast<void*>(card),
                .read = read,
                .prog = prog,
                .erase = erase,
                .sync = sync,
            #ifdef LFS_THREADSAFE
                .lock = nullptr,
                .unlock = nullptr,
            #endif
                .read_size = block_size,
                .prog_size = block_size,
                .block_size = block_size,
                .block_count = block_count,
                .block_cycles = -1,
[... 279 lines cut ...]
            return ESP_FAIL;
        }

        static std::array<uint8_t, urazr.size()> read_buf;
        read_buf.fill(0);
        if(sdmmc_read_sectors(&card, read_buf.data(), done->value(), 1) != ESP_OK) {
            std::printf("sdmmc_read_sectors(&card, read_buf.data(), %zu, 1) != ESP_OK)\n", done->value());
            done->reset();
            return ESP_FAIL;
        }

        if(read_buf != urazr) {
            std::printf("read_buf != urazr, i: %zu\n", done->value());
            done->reset();
            return ESP_FAIL;
        }
    }

    done->reset();
    return ESP_OK;
}
```
And it's super slow, we're talking 128 sectors/s ~ 128*512 bytes/s. But it seems to work. It's supposed to take 9 hours to complete?

Honestly, I don't know what I'm doing and I'm running out of ideas.

huming2207, 2024-03-09:
> Although this code seems to work at first, it all falls apart when trying to use it to create larger files or multiple more than 100 files with size of 512 bytes. The occurence of the failures seems [...]

Hmmm that sounds like a low-level issue to me, it could be Espressif's SPI and SD card driver or hardware connections (e.g. wiring is too long). 

I've also seen occasionally some SD card over SPI may behave differently compared to SD over SDMMC mode. I remembered I've read a Chinese article long time ago that someone managed to recover a semi-bricked SD card data with lower-speed SPI, while he couldn't read out anything by normal USB card readers over SDMMC. Maybe in your case this could be somewhat similar? Have you tried any other SD cards? Particularly some newer ones?

BrianPugh, 2024-06-03:
closing due to lack of activity.

closed 2024-06-03 by BrianPugh