joltwallet/esp_littlefs #192: littlefs_python-0.11.0 causes build error on esp-idf

AxelLin, 2024-07-05:
I'm using esp-idf v5.2.2-272-gabf8ffac59
joltwallet/littlefs (1.14.6)

I notice the littlefs-python is upgraded:
[398/1094] Generating ../../littlefs_py_venv
Ignoring littlefs-python: markers 'python_version < "3.8"' don't match your environment
Collecting littlefs-python>=0.7.1
  Downloading littlefs_python-0.11.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (946 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 946.4/946.4 KB 1.6 MB/s eta 0:00:00
Installing collected packages: littlefs-python
Successfully installed littlefs-python-0.11.0

And then I got below build error:
```
/home/axel/esp/esp-idf-dev/apps/my-proj/build/littlefs_py_venv/bin/littlefs-python create /home/axel/esp/esp-idf-dev/apps/my-proj/resources -v --image=/home/axel/esp/esp-idf-dev/apps/my-proj/build/res.bin --fs-size=0x4e000 --name-max=64 --block-size=4096
usage: littlefs-python [-h] [--version] {create,extract,list} ...
littlefs-python: error: unrecognized arguments: --image=/home/axel/esp/esp-idf-dev/apps/my-proj/build/res.bin
```

seems the argument is changed? (--image is is no longer exist?)
```
$ build/littlefs_py_venv/bin/littlefs-python create -h
usage: littlefs-python create [-h] [-v] [--name-max NAME_MAX] --block-size BLOCK_SIZE (--block-count BLOCK_COUNT | --fs-size FS_SIZE) source [destination]

positional arguments:
  source                Source file/directory-of-files to encode into a littlefs filesystem.
  destination           Output LittleFS filesystem binary image.

options:
  -h, --help            show this help message and exit
  -v, --verbose
  --name-max NAME_MAX   LittleFS max file path length. Defaults to LittleFS's default (255).
  --block-size BLOCK_SIZE
                        LittleFS block size.
  --block-count BLOCK_COUNT
                        LittleFS block count
  --fs-size FS_SIZE     LittleFS filesystem size. Accepts byte units; e.g. 1MB and 1048576 are equivalent.
```

BrianPugh, 2024-07-05:
Can you try upgrading to `esp_littlefs` v1.14.7. I added compatibility with littlefs-python v0.11.0 cli.

BrianPugh, 2024-07-05:
ugh sorry, v1.14.7 is still having issues, I'll release a fixed 1.14.8 soon.

BrianPugh, 2024-07-05:
please give v1.14.8 a try now.

finger563, 2024-07-05:
@BrianPugh fixed it for me, thanks for the rapid fix! 🎊 
https://github.com/esp-cpp/espp/pull/275

BrianPugh, 2024-07-05:
excellent, I'll assume that this fixes it for others (it should!). Sorry for the original breaking build!

closed 2024-07-05 by BrianPugh