joltwallet/esp_littlefs #193: littlefs_create_partition_image does not create the binary

AndresSingularThings, 2024-07-05:
I’m using ESP-IDF v5.2.2-272-gabf8ffac59 and joltwallet/littlefs (1.14.7).

The issue I’m facing is that the partition binary is not being created. When I attempt to flash, I encounter the following error: `write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'prv.bin'.`

In the CMakeLists.txt file, I have the following: `littlefs_create_partition_image(prv lfs_private_content FLASH_IN_PROJECT).`

I’m not sure what happened. In version 1.14.4, it worked fine, but now when I downgrade, I encounter the same problem.

dovydas1928, 2024-07-05:
we are having the same issue, waiting v1.14.8

BrianPugh, 2024-07-05:
please give v1.14.8 a try now.

dovydas1928, 2024-07-05:
perfect. now works. thanks!

BrianPugh, 2024-07-05:
great! sorry for the original breaking build! glad it works now!

closed 2024-07-05 by BrianPugh