joltwallet/esp_littlefs #196: Support for "generic" attributes

josesimoes, 2024-07-23:
Any chance to add an API to get/set attributes? 
A simple byte will suffice to hold the simple (and usual ones) like hidden, read-only, system.
littlefs has this and I've noticed this already being used to store a time stamp.

BrianPugh, 2024-07-23:
I haven't had too much of a chance to think about this, but it might be kind of complicated to expose. Lets say we want two functions:

```
esp_err_t esp_littlefs_getattr(const char *path, uint8_t type, void *buffer, size_t size);
esp_err_t esp_littlefs_setattr(const char *path, uint8_t type, void *buffer, size_t size);
```

We need to get the `esp_littlefs_t` context from the path. Normally, this is handled by esp-idf's VFS and passed along to a bunch of hooks that we register. Just skimming [the vfs page](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/vfs.html), I don't see a function that goes from `path` to `context`. This is to:

1. Get the main `lfs` object.
2. Get the correct mutexs so it's all threadsafe.
3. Get the correct path (vfs removes the mounting point prefix for you).

So if we have a way to do that, I can add this functionality.

BrianPugh, 2024-07-23:
There are the private [`get_vfs_for_path`](https://github.com/espressif/esp-idf/blob/af25eb447e3330c21e3b38e91db16332056882b2/components/vfs/vfs.c#L361) and [`translate_path`](https://github.com/espressif/esp-idf/blob/af25eb447e3330c21e3b38e91db16332056882b2/components/vfs/vfs.c#L351) functions that do exactly what we need, but I'd be a bit hesitant to reach in and use those.

josesimoes, 2024-07-25:
@BrianPugh that looks like a good path forward. Understood those are private, still... 😉 
Following this I took the liberty to suggest adding API to get/set attributes in VFS. Please follow the issue here espressif/esp-idf#14248.