joltwallet/esp_littlefs #199: Invalid WARNING emssage

ammaree, 2024-08-07:


ammaree, 2024-08-07:
For some time, possibly last 12 months, we have been receiving a message at warning level.

'vfs esp_vfs_register_fd_range is successful for range <54; 64) and VFS ID 5'

Surely this message should be output at debug or verbose level?

BrianPugh, 2024-08-07:
This warning comes from upstream esp-idf [here](https://github.com/espressif/esp-idf/blob/8e4454b285ad39881c5bf3f440d8be617a2f18a8/components/vfs/vfs.c#L161). I agree it should probably be at a lower logging level.

closed 2024-08-07 by BrianPugh

ammaree, 2024-08-07:
How do we resolve this?
Has it been raised with them?

ammaree, 2024-08-07:
Just raised it there, could not find a previous issue based on a quick search.

Thanks

BrianPugh, 2024-08-07:
linking: https://github.com/espressif/esp-idf/issues/14327