joltwallet/esp_littlefs #202: Adding CycloneDx JSON into project

3100jisa, 2024-10-18:
Hello there!
I see file library.json in the root of your project which partially corresponds to CycloneDx format (I am using version 1.5) - see https://cyclonedx.org/docs/1.5/json/.
What is mainly missing for me there is CPE number (see https://nvd.nist.gov/products/cpe/).
Are you interested in this topic? Obtaining CPE number, integrate proper CycloneDx file (no need to have name library.json for it)
and potentially get your project integration monitored by tools like Vigiles (https://www.timesys.com/solutions/vigiles-vulnerability-management/).

Thanks for your response.

Regards
Jiří

BrianPugh, 2024-10-18:
I'm not familiar with CPE, but am will to accept instructions/PR for integrations.

3100jisa, 2024-10-21:
I am not allowed to help with CPE (someone needs to request that by mailing NIST) but I will open PR with made up CPE according to expectation based on experience.
CPE line in provided JSON shall look like this: "cpe": "cpe:2.3:a:joltwallet:esp_littlefs:1.14.8:*:*:*:*:*:*:*".

I will propose name of the file like cyclonedx.json or so. So it will not colide with your remove file library.json.

BrianPugh, 2024-10-21:
At the current time I'm not looking to support this, but if more people need this we can re-investigating support. Thanks!

closed 2024-10-21 by BrianPugh