joltwallet/esp_littlefs #208: After the second encryption and burning, the data in the file system will be cleared.

wolen666, 2024-12-19:
I built the littlefs file system under esp-matter and then performed the encryption operation. After the first encryption was successful, there were no problems with the file system. However, when I then used idf.py encrypted-flash monitor, all the files in the file system disappeared. 
The main steps are as follows
step1:
    # Name,   Type, SubType, Offset,   Size,     Flags
    nvs,      data, nvs,     ,        0x8000,
    fctry1,   data, nvs_keys,,        0x1000,   encrypted
    otadata,  data, ota,     ,        0x2000,
    phy_init, data, phy,     ,        0x2000,
    ota_0,    app,  ota_0,   ,        0x1B2000,
    ota_1,    app,  ota_1,   ,        0x1B2000,
    storage,  data, spiffs,,        0x68000,  
    fctry,    data, nvs,     ,        0x6000,
[image]

   littlefs_create_partition_image(storage ../html  FLASH_IN_PROJECT)

When I did not encrypt, the file system was normal no matter how many times I flashed it. When I encrypted it for the first time, there was no problem. Then when I encrypted and wrote it for the second time, all the files under the html were gone.
The following is the log after the device is started.

LittleFS partition size: total: 425984, used: 8192
Listing files in: /html

esp-idf  commit e3e112e6122dfd064a78f3dece8cdce17a472e13 (HEAD -> release/v5.1)

wolen666, 2024-12-20:
I am also very confused about this issue. If I use the spiffs file system, there will be no problem in flashing again.

BrianPugh, 2025-02-06:
closing due to inactivity.

closed 2025-02-06 by BrianPugh