joltwallet/esp_littlefs #209: ESP32-S3 - Data from USB CDC gets randomly corrupted

TheCrypt0, 2024-12-22:
Hi, thank you for the great work on this module!

I've just implemented it on our project but we are facing random corruption of the data when writing to the filesystem.

Our communication system isn't super trivial but I'll make my best to explain it.

We use the esp_timer to check every 10ms if a json rpc message has been received via USB CDC, if so, we parse and pass it to the relevant handler. The data is in base64 and then decoded runtime by the function below.

The flow is as follows:
1. Delete the old file using an rpc call (that uses `unlink`)
2. Send the new file in chunks that they get appended to the file until we completed the transfer.
3. We use the file on the embedded system.

The weird thing that is happening is that after some time (or event, I honestly don't know), every file we send (and write to littlefs) gets corrupted and only a complete format of the partition fixes it. After some time it starts to happen again. 
By corrupted I mean shifted, the data seems ok but its completely shifted all over the place with the wrong offsets.

The partition is 2MiB and currently it just has two files: `file.json ~16K` and `file.bin ~80K`.

The firmware and the pc software remain the same, the only thing that changes is the littlefs partition that gets formatted.

During those tests I've also re-flashed the firmware but the littlefs partition isn't included in the image.

*Do you have any idea on what could be causing this or what steps I could try to troubleshoot the issue?* Thank you in advance.

Function that writes from json rpc to flash:

```cpp
void
fs::on_fs_writebin(const rpc* handler, const JsonDocument& req)
{
  JsonDocument doc;
  auto         params = handler->get_params(req);

  if (!params["data"]) {
    send_error_missing_param(handler, req, "data");
    return;
  }

  size_t offset = params["offset"] | 0;
  bool   append = params["append"] | false;

  auto  file_name = fix_filename(params["file"]);
  FILE* file      = fopen(file_name.c_str(), append ? FILE_APPEND : FILE_WRITE);
  if (!file) {
    send_error_file_open(handler, req);
    return;
  }

  if (!append && offset > 0) {
    if (!seek_file(file, offset, handler, req))
      return;
  }

  const char* base64_data     = params["data"].as<const char*>();
  size_t      input_len       = strlen(base64_data);
  size_t      max_decoded_len = input_len / 4 * 3 + 1;

  wl_buffer<uint8_t> decoded_data(max_decoded_len);
  if (!decoded_data) {
    fclose(file);
    send_error_memory(handler, req, "decoded data");
    return;
  }

  size_t decoded_len;
  if (mbedtls_base64_decode(decoded_data.get(),
                            max_decoded_len,
                            &decoded_len,
                            (const unsigned char*)base64_data,
                            input_len) != 0) {
    fclose(file);
    WL_ERROR("Failed to decode base64 data");
    handler->send_error(req, -1, "Failed to decode base64 data");
    return;
  }

  size_t bytes_written = fwrite(decoded_data.get(), 1, decoded_len, file);

  // Ensure data is flushed to flash
  fflush(file);
  fsync(fileno(file));

  // Close the file
  fclose(file);

  if (bytes_written != decoded_len) {
    WL_ERROR("Failed to write file");
    handler->send_error(req, -1, "Failed to write file");
    return;
  }

  doc["bytes_written"] = bytes_written;
  doc["offset"]        = offset;
  handler->send_response(req, doc);

  notify_file_write(file_name);
}
```

BrianPugh, 2024-12-22:
Hello! I can try and look at this more closely tomorrow. In the meantime, what version of esp_littlefs are you running? I just saw that Geky released a (possibly unrelated) bugfix in the littlefs core codebase. I'll release a patch release updating the littlefs submodule to that tomorrow.

TheCrypt0, 2024-12-22:
Hi, thank you for the hyper-fast response! I'm using the latest stable `1.16.0`.

Now that you have suggested it, I manually upgraded the submodule `littlefs` on my project. I saw this issue https://github.com/littlefs-project/littlefs/issues/1057 that could possibly be traced back to what I was experiencing too.

BrianPugh, 2024-12-22:
what might also be helpful is turning on verbose-level logs for esp_littlefs:

```
esp_log_level_set("littlefs", ESP_LOG_INFO);  # or is it esp_littlefs? it's been a while for me --'
```

TheCrypt0, 2024-12-23:
The last update has fixed the issue. Thank you for the support!

closed 2024-12-23 by TheCrypt0