joltwallet/esp_littlefs #213: Watchog in while loop in fwrite() on full filesystem.

ddomnik, 2025-01-10:
Usually on a full filesystem littleFS returns from fwrite() within a few ms and prints:
E (2818207) esp_littlefs: ./managed_components/joltwallet__littlefs/src/littlefs/lfs.c:689:error: No more free space 0x48d

However, it could happen that the library gets "stuck" for a very long time (multiple seconds or even minutes - prob. depending on partition size). This happens at exactly in this while loop. (The second `while` in lfs_alloc())
https://github.com/littlefs-project/littlefs/blob/d01280e64934a09ba16cac60cf9d3a37e228bb66/lfs.c#L656

The problem is, that usually every program have watchdogs. Espressifs default watchdog has a maximum of 60 seconds, which seems to be insufficient in rare cases. [maximum wdt timeout](https://docs.espressif.com/projects/esp-idf/en/v5.1.4/esp32/api-reference/kconfig.html#config-esp-task-wdt-timeout-s) 

The filesystem size is 7080K. Espressif IDF Version 5.0.6 and 
  joltwallet/littlefs:
    component_hash: 362f1f5beb5087b0c60169aff82676d2d0ffc991ead975212b0cba95959181c5
    source:
      type: service
    version: 1.14.8

Maybe some experts know how to handle this issue. 

Based on my current understanding of the issue, a very simple thought would be to give littleFs it's own watchdog that can be set via the menuconfig. This allows to "catch" the issue on application level rather than system level.

BrianPugh, 2025-01-11:
Long hangs are certainly not good. However, I don't know if any action can be taken in this codebase. AFAIK the closest-to-appropriate WDT in esp-idf is the task WDT, but that would be something the developer has to define at the task level, and thusly must be external to esp_littlefs. Totally open to additional ideas/features that improve usability!

BrianPugh, 2025-02-03:
We can re-open if any actionable items come up.

closed 2025-02-03 by BrianPugh