joltwallet/esp_littlefs #214: Performance issues of esp_littlefs

yiming564, 2025-01-18:
Created a seperated issue from #211 .

In short, esp_littlefs's sdmmc speed is pretty slow, and from [my simple test](https://github.com/joltwallet/esp_littlefs/issues/211#issuecomment-2575585456) the sequential writing speed at 40Mhz 4 line sdmmc mode is **48 KiB/s**.

In conparison, [FatFS's sequential writing speed](https://github.com/joltwallet/esp_littlefs/issues/211#issuecomment-2598637311) is **13.76 MiB/s**, hundreds of times of esp_littlefs's speed.

---

I do all these tests on a U3 (UHS Speed Class 3) SD Card so it's unlikely to be the bottleneck, the ASSSD Benchmark score is shown below:

[image]

---

LittleFS's configuration:

```
CONFIG_LITTLEFS_SDMMC_SUPPORT=y
CONFIG_LITTLEFS_MAX_PARTITIONS=3
CONFIG_LITTLEFS_PAGE_SIZE=512
CONFIG_LITTLEFS_OBJ_NAME_LEN=255
CONFIG_LITTLEFS_READ_SIZE=128
CONFIG_LITTLEFS_WRITE_SIZE=128
CONFIG_LITTLEFS_LOOKAHEAD_SIZE=512
CONFIG_LITTLEFS_CACHE_SIZE=512
CONFIG_LITTLEFS_BLOCK_CYCLES=512
CONFIG_LITTLEFS_USE_MTIME=y
# CONFIG_LITTLEFS_USE_ONLY_HASH is not set
CONFIG_LITTLEFS_HUMAN_READABLE=y
CONFIG_LITTLEFS_MTIME_USE_SECONDS=y
# CONFIG_LITTLEFS_MTIME_USE_NONCE is not set
# CONFIG_LITTLEFS_SPIFFS_COMPAT is not set
# CONFIG_LITTLEFS_FLUSH_FILE_EVERY_WRITE is not set
CONFIG_LITTLEFS_FCNTL_GET_PATH=y
CONFIG_LITTLEFS_FCNTL_F_GETPATH_VALUE=20
CONFIG_LITTLEFS_MULTIVERSION=y
CONFIG_LITTLEFS_DISK_VERSION_MOST_RECENT=y
# CONFIG_LITTLEFS_DISK_VERSION_2_1 is not set
# CONFIG_LITTLEFS_DISK_VERSION_2_0 is not set
# CONFIG_LITTLEFS_MALLOC_STRATEGY_DISABLE is not set
CONFIG_LITTLEFS_MALLOC_STRATEGY_DEFAULT=y
# CONFIG_LITTLEFS_MALLOC_STRATEGY_INTERNAL is not set
CONFIG_LITTLEFS_ASSERTS=y
```

yiming564, 2025-01-19:
More info: LittleFS vs FATFS on spi flash:

```
I (00:00:03.215) app_manager: Start stress test /sys/BigFile.txt.
I (00:00:03.276) app_manager: Write 0/64 blk.
I (00:00:43.954) app_manager: Completed!
I (00:00:43.956) app_manager: File size check: 4194304 Bytes
I (00:00:43.956) app_manager: Start read test.
I (00:00:43.960) app_manager: Read 0/64 blk.
I (00:00:44.508) app_manager: Completed!
I (00:00:44.509) app_manager: File size check: 4194304 Bytes
I (00:00:44.509) app_manager: Start stress test /fat/BigFile.txt.
I (00:00:44.516) app_manager: Write 0/64 blk.
I (00:01:50.833) app_manager: Completed!
I (00:01:51.020) app_manager: File size check: 4194304 Bytes
I (00:01:51.020) app_manager: Start read test.
I (00:01:51.021) app_manager: Read 0/64 blk.
I (00:01:51.335) app_manager: Completed!
I (00:01:51.336) app_manager: File size check: 4194304 Bytes
```

||Read|Write|
|:-:|:-:|:-:|
|LittleFS|7.299 MiB/s|0.098 MiB/s|
|FATFS|12.73 MiB/s|0.086 MiB/s|

spi flash's configuration is QIO 120 Mhz.

BrianPugh, 2025-01-19:
That's an excellent datapoint, and about the performance difference when compared to FAT that I would expect. This strongly indicates that it's an sdmmc specific issue, probably related to read/write sizes.