joltwallet/esp_littlefs #216: Build warning calloc-transposed-args

mwaldoch, 2025-01-30:
Building apps that use esp_littlefs with ESP-IDF v5.4 is giving build warnings about invalid order of arguments for `calloc`. I have to admit that I'm on an older version of esp_littlefs, but I see that the problem will probably be the same on the newest commit [here](https://github.com/joltwallet/esp_littlefs/blob/master/src/esp_littlefs.c#L282) and [here](https://github.com/joltwallet/esp_littlefs/blob/master/src/esp_littlefs.c#L1151)

That is unless the compiler is mistaken. I suppose it works since `a * b` is equal to `b * a`

Compiler output:
```
[1365/1413] Building C object esp_littlefs/src/esp_littlefs.c: In function 'esp_littlefs_format':
esp_littlefs/src/esp_littlefs.c:393:35: warning: 'calloc' sizes specified with 'sizeof' in the earlier argument and not in the later argument [-Wcalloc-transposed-args]
  393 |         efs->cache = calloc(sizeof(*efs->cache), efs->cache_size);
      |                                   ^
esp_littlefs/src/esp_littlefs.c:393:35: note: earlier argument should specify number of elements, later size of each element

esp_littlefs/src/esp_littlefs.c: In function 'esp_littlefs_init':
esp_littlefs/src/esp_littlefs.c:724:35: warning: 'calloc' sizes specified with 'sizeof' in the earlier argument and not in the later argument [-Wcalloc-transposed-args]
  724 |         efs->cache = calloc(sizeof(*efs->cache), efs->cache_size);
      |                                   ^
esp_littlefs/src/esp_littlefs.c:724:35: note: earlier argument should specify number of elements, later size of each element
```

BrianPugh, 2025-01-30:
Thanks for posting this! Fixing now!

BrianPugh, 2025-01-30:
addressed in v1.16.3!

closed 2025-01-30 by BrianPugh