joltwallet/esp_littlefs #220: WARNING: Partition has name 'littlefs' which is a partition subtype, but this partition has non-matching type 0x1 and subtype 0x82. Mistake in partition table?

diplfranzhoepfinger, 2025-02-15:
Can this be ? 

WARNING: Partition has name 'littlefs' which is a partition subtype, but this partition has non-matching type 0x1 and subtype 0x82. Mistake in partition table?

Thanks, 
Franz

diplfranzhoepfinger, 2025-02-15:
https://github.com/espressif/esp-idf/blob/master/examples/storage/littlefs/partitions_demo_esp_littlefs.csv

`storage,  data, littlefs,      ,  0xF0000,`

diplfranzhoepfinger, 2025-02-15:
https://github.com/joltwallet/esp_littlefs/blob/71e549277d3c728a87281b843f654638cb5eb154/example/partitions_demo_esp_littlefs.csv

`littlefs,  data, spiffs,      ,  0xF0000, `

unclear !

BrianPugh, 2025-02-15:
Thanks for bringing this up! I didn't realize that esp-idf added `littlefs` as a subtype. It seems like this exists for esp-idf >=v5.0. Since esp-idf v4.4 EOL was July 2024, I think we can safely merge in your PR #221 and #222. Thanks!

closed 2025-02-15 by BrianPugh