joltwallet/esp_littlefs #225: Calculating the occupied space

butaikis, 2025-02-21:
Hi. There are 3 files in my storage, their total size is **65153** bytes, but `FS_User.usedBytes()` shows **73728**. Can you explain how the size of the occupied space is calculated and what value is relevant?

BrianPugh, 2025-02-21:
internally, it calls `lfs_fs_size`, who's docstring is [here](https://github.com/littlefs-project/littlefs/blob/0494ce7169f06a734a7bd7585f49a9fa91fa7318/lfs.h#L724). I copied/pasted it below:

```C
// Finds the current size of the filesystem
//
// Note: Result is best effort. If files share COW structures, the returned
// size may be larger than the filesystem actually is.
//
// Returns the number of allocated blocks, or a negative error code on failure.
lfs_ssize_t lfs_fs_size(lfs_t *lfs);
```

If you'll notice, 73728 is a multiple of 4096 (the block size): `4096 * 18 = 73728`. So 18 blocks are in use in your filesystem, but they might not be 100% filled up. There's also some storage overhead in maintaining metadata about the filesystem and its state.

Some interesting related threads:
* https://github.com/littlefs-project/littlefs/issues/927
* https://github.com/littlefs-project/littlefs/issues/840
* https://github.com/littlefs-project/littlefs/issues/980#issuecomment-2106913268
  > Because littlefs uses copy-on-write data-structures, the result may sometimes be larger than the actual usage, but it should never be smaller. Hopefully it's still useful for estimating how full the disk is.

butaikis, 2025-02-21:
@BrianPugh Thanks, I looked at the links. But I still don't understand, should I rely on usedBytes() when calculating free space, or should I just calculate the actual size of the files?

BrianPugh, 2025-02-21:
It depends on your use-case, but generally:
```
MIN(FS_User.usedBytes(), total_storage_area);
```
is a good indicator for most uses (seeing if there's enough room for another file, displaying filesystem utilization to the user, etc). It's slightly conservative, but that's better than the other way around.

butaikis, 2025-02-22:
@BrianPugh so in any case, it will return `FS_User.usedBytes()`, or not?

BrianPugh, 2025-02-22:
Correct

closed 2025-02-28 by BrianPugh