joltwallet/esp_littlefs #227: Function stat if used on folders returns strange values.

uqfus, 2025-02-26:
When stat function used on file it returns normal values:
`>ls /littlefs
<type> <size> <mtime> <name>
file    1980 2025.02.02 23:43:00 /littlefs/charger_172650024173.chart
file    2600 2025.02.02 19:37:00 /littlefs/charger_183371037950.chart
file   12033 1970.01.01 03:00:00 /littlefs/master.log.1
file     979 2025.02.26 17:30:20 /littlefs/master.log
file   12048 2025.02.25 23:31:08 /littlefs/test.log`
When stat function used on directory it returns this:
`dir  1073539712 1970.01.01 02:59:59 /littlefs/test2`
size? 1073539712 0x3FFCEA80 
mtime? uint64_t = -1

BrianPugh, 2025-02-26:
good find! definitely a bug. #228

uqfus, 2025-02-26:
About mtime.
vfs_littlefs_rename
vfs_littlefs_mkdir
also should update mtime

my vote is update mtime on close
vfs_littlefs_open
#if CONFIG_LITTLEFS_USE_MTIME
    if (lfs_flags != LFS_O_RDONLY) {
        /* If this is being opened as not read-only */
        vfs_littlefs_update_mtime(efs, path);
    }
#endif

vfs_littlefs_close(void* ctx, int fd) {
    // TODO update mtime on close? SPIFFS doesn't do this

BrianPugh, 2025-02-27:
the original issue (st_size of directories) has been fixed in v0.17.0. We can continue mtime discussion in #229

closed 2025-02-27 by BrianPugh