joltwallet/esp_littlefs #234: Writing file data out of order causes file corruption

fettinn, 2025-04-03:
Using release version 1.19.1 / 2.10.2
ESP IDF version 5.1.3
chip esp32s3

Our legacy libraries format a file in the following way
2 bytes header
32 bytes crc/hash
+variable length of text data

Therefore, when writing these files, the data is always written first
Sequence of events

```
fopen
fseek 34
fwrite
fclose
...
fopen
fseek 34
fread (read all data to compute hash)
fclose
...    [Verified everything is correct up to this point]
fopen
fseek 0
fwrite 34 bytes
fclose
```

Upon subsequent open and reads, on the first 34 bytes are valid, the following data is junk. Based on the reads I have performed after, it appears the file is only 34 bytes in length after the final write, as if the following data in the file cache is invalid.

Here is the log, I have removed sensitive information
Data after >> is what is being written/read
All zeroes (\0) in the data have been replaced with '~' for readability.

```
Opening /file.tmp
W (5447) esp_littlefs: write addr 0134ed80, size 00000080, err 0
W (5447) esp_littlefs: write addr 0134ee00, size 00000080, err 0
Seeking to offset 34
Writing /file.tmp: 9 bytes
>>bootups=1
Fsync
W (5457) esp_littlefs: write addr 0134ee80, size 00000080, err 0
Closing
Opening /file.tmp
Seeking to offset 34
Reading /file: 127 bytes
>>bootups=1~~~~~~~~~~~~~~~~~~~~~~~~~bootups=1~~~~~~@~~b���~~~~A~~~~~~~~~~~@~~~~~~~z���~~~~@~~@~@~~~@~~~D~TA~r��P~
Closing
Opening /file.tmp
W (5567) esp_littlefs: write addr 0134ef00, size 00000080, err 0
Seeking to offset 0
Writing /common/file.tmp : 34 bytes
>>P*12345678901234567890123456789012
Fsync
W (5567) esp_littlefs: write addr 0134ef80, size 00000080, err 0
Closing
Renaming /file.tmp to /file.txt
W (5917) esp_littlefs: write addr 0134d000, size 00000200, err 0
W (5997) esp_littlefs: write addr 0134d200, size 00000080, err 0
Opening /file.txt
W (5997) esp_littlefs: write addr 0134d280, size 00000080, err 0
Reading /file.txt : 127 bytes
>>P*12345678901234567890123456789012~pEEb���1x5yCbu476WeeW58u11qedz���4pu50xEfUuU1859b9B5t75Uer���05q08Ef5TUU45Ueuu8
E (6227) [task]: Data integrity failed for /file.txt!
Closing 0
```

It is also worth noting that the above worked in version 1.6.0, however we were having issues when multiple fclose operations were injected with multi task writing with quick fopen/fclose operations. They couldnt be ignored because the first fclose would happen earlier than expected, resulting in half written data.
The code does so many open and closes because it is tied to streams that go in and out of scope. The work required to optimize is very high.

closed 2025-04-04 by fettinn

BrianPugh, 2025-04-04:
@fettinn were you able to solve your issue?

fettinn, 2025-06-10:
Issue is solved. It was not an issue with this repo