joltwallet/esp_littlefs #237: error: implicit declaration of function 'esp_random'

nopnop2002, 2025-06-26:
When CONFIG_LITTLEFS_MTIME_USE_NONCE is enabled, we get this error.

```
/home/nop/esp-idf/examples/storage/littlefs/managed_components/joltwallet__littlefs/src/esp_littlefs.c: In function 'vfs_littlefs_utime':
/home/nop/esp-idf/examples/storage/littlefs/managed_components/joltwallet__littlefs/src/esp_littlefs.c:2359:26: error: implicit declaration of function 'esp_random'; did you mean 'srandom'? [-Wimplicit-function-declaration]
 2359 |         if( 0 == t ) t = esp_random();
      |                          ^~~~~~~~~~
      |                          srandom
[507/515] Building C object esp-idf/joltwallet__littlefs/CMakeFiles/__idf_joltwallet__littlefs.dir/src/littlefs/lfs.c.obj
ninja: build stopped: subcommand failed.
HINT: esp_random.h header file is not included by esp_system.h anymore. It shall then be manually included with #include "esp_random.h"
ninja failed with exit code 1, output of the command is in the /home/nop/esp-idf/examples/storage/littlefs/build/log/idf_py_stderr_output_21984 and /home/nop/esp-idf/examples/storage/littlefs/build/log/idf_py_stdout_output_21984
```

[image]

BrianPugh, 2025-06-27:
hm, very odd. We're just using this [function/header](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/random.html#header-file). It all compiles fine on my machine.

nopnop2002, 2025-06-27:
@BrianPugh 

This problem only occurs when CONFIG_LITTLEFS_MTIME_USE_NONCE=y.

nopnop2002, 2025-06-27:
I found the reason.

It was because the version of idf_component.yml I was using was old.

Once I updated the version of idf_component.yml, I was able to build correctly.

```
## IDF Component Manager Manifest File
dependencies:
  joltwallet/littlefs: "~=1.20.0"
```

Sorry for the trouble.

closed 2025-06-27 by nopnop2002